hneemann/Digital #428: Dialog windows don't work on second monitor

dfbg, 2020-03-23:
Drag a circuit window to a second monitor that is to the left of the main monitor.
Open a dialog window like right-click component attributes or Circuit Specific Settings.

Sometimes the window is completely blank.

When the contents are present most controls are non-functional.  Most drop-downs don't drop.  Text fields have no cursor and take no typing.  [Open Circuit] opens a new window but it's blank.

The [Help]/[Cancel]/[OK] buttons work fine.

Move the window back to the main window and everything works fine again.

MacOS 10.14.6
Java 8.211

hneemann, 2020-03-24:
There is no code in the simulator that does anything special if two monitors are used.
I think this is a problem of the JVM or Swing libraries on the Mac.

The only thing where multiple monitors are considered is when determining the position of the window. Does the window appear at the expected position, or is it also placed completely wrong?

dfbg, 2020-03-24:
The windows were all in the right places so your code seems fine.

After an unrelated reboot of the system this morning it's working as expected now.  It may never happen again.

closed 2020-03-24 by dfbg