hneemann/Digital #429: 3-state bus doesn't support pull-up/down resistors

dfbg, 2020-03-24:
When a 3-state bus has all connected outputs in Z and also has pull-down resistors the wire is drawn grey rather than dark green.

I tried switching one pull-down to a pull-up and the results on the bus input device didn't change.  So Z defaults to low?  Do I need to break out the bus wires individually?

hneemann, 2020-03-24:
High Z is supported for individual bits in a bus. As soon as a single bit is in the High Z state, the wire is colored gray.

[image]

What can be confusing is that splitters are unidirectional. See #197

dfbg, 2020-03-24:
I flipped the splitter around so the pull-downs are on the input side and now I get a Short Circuit error.  The pull-down behavior seems to be lost going through the splitter.  I guess I'll have to use individual wires.

hneemann, 2020-03-24:
The pull downs only work on the input side of a splitter. The output side is driven low if a pull down is connected to the input side.
It is best to think of the splitter as an active bus driver that has zero propagation delay.

dfbg, 2020-03-24:
I solved my circuit by connecting a pull-down directly to the bus wire and setting it to 8 bits.

closed 2020-03-24 by dfbg