hneemann/Digital #430: How to run the MIPS example on Basys3?

suluoniu, 2020-03-26:
can the MIPS CPU example  be downloaded to the basys3 ?

hneemann, 2020-03-26:
To run the processor example on a Basys3 board three steps are required:
1. At first select the BASYS3 config in the tool-chain settings. The file is called `BASYS3.config`
   This allows the simulator to add the FPGA specific hdl code required to create the clock signal.
2. Export a HDL file (either VHDL or Verilog) using the new BASYS3 menu items.
    This also creates a Vivado project which can be opened directly by Vivado.
3. Use Vivado to create the bit stream and to flash the bit stream to the board.

suluoniu, 2020-03-26:
I do as you told me, only the example ProcessorMux.dig can export vhdl successfully ,is it right?

hneemann, 2020-03-26:
Yes, there are two examples available: `HDLExample.dig` and `VGAExample.dig`.
In `HDLExample.dig` the processor runs a program that creates a lightchase that can be controlled with the buttons. And `VGAExample.dig` also contains a VGA card and the running program displays a character set on a connected VGA monitor and you can move a smiley across the screen with the buttons.

Both examples use ProcessorMux.dig because it uses a muxer instead of a High-Z bus. Bus drivers with High-Z output are not supported in HDL export. See also [here](https://github.com/hneemann/Digital/issues/184#issuecomment-416529591).

closed 2020-04-02 by hneemann