hneemann/Digital #431: Auto-increment tunnels when pasted.

bcrew1375, 2020-03-31:
There is a very useful feature for inputs and outputs that when copying and pasting an already named component, the numbers will be incremented automatically, which immensely reduces the tedium and risk of mistakes when renaming a large number of them. This feature doesn't seem to exist for tunnels though. I'm using tunnels for a very large number of inputs and outputs in one circuit that all relate to the same thing, but I have to go through and change the numbers for each of them. It would be very nice to have an auto-increment as the input and output pasting does.

hneemann, 2020-03-31:
Tunnels work slightly differently compared to inputs and outputs. This is because you always need several tunnels with the same name, which is not the case with inputs and outputs.
However, if a tunnel is inserted and it is named, the next directly inserted tunnels (hotkey [T]) are automatically named. If these are then copied, the names are preserved.

bcrew1375, 2020-03-31:
So, I tried with directly inserting the tunnels. They do indeed increment if directly inserted with the T hotkey, but if I copy and paste them, the numbers don't change. I'm not sure if I misunderstood or did something wrong.

hneemann, 2020-04-01:
When you copy and paste tunnels, the net names are intentionally not changed. This is because you often want to copy tunnels without changing the name, because you always need several tunnels with the same net name.
If the net name of a tunnel were unique, the tunnel would make no sense. For an input or output, it is the opposite.

bcrew1375, 2020-04-01:
Right, you need to have an entry point and exit point for a tunnel. So, there's no way to create a large array of related tunnels without inserting them individually using the T hotkey? Like if I want to create a set of tunnels that direct 64 input signals to a circuit that takes 64 inputs and have tunnels labeled as input1, input2, input3, etc., I would need to insert each tunnel individually using the hotkey?

hneemann, 2020-04-01:
Yes, you have to create the 64 tunnels used as entry points with the hotkey [T], which takes about one minute. The 64 tunnels for the exit points can then be created using copy&paste.
And since there is often more than one exit point, all of them can be created by copy&paste.

But maybe you should use a bus instead of 64 tunnels.

bcrew1375, 2020-04-01:
You're probably right. I'm attempting to build a CPU from the basic logic gates up following the nand2tetris course, and this is the first exposure I've had to a lot of this. I'm constructing an 8-way 16-bit multiplexer. I've already made a 4-way one and using a bus probably would've saved me a lot of work.

hneemann, 2020-04-01:
The nand2tetris course is without doubt worth getting into.
However, if you have built and understood your own multiplexer, it's okay to use the built-in multiplexer in the further process. 
Once you understand how a multiplexer works, you won't learn much more if you build a 16-bit 8-way multiplexer by hand. It only causes very tedious and error-prone work. And I think that goes for all components.

closed 2020-04-04 by hneemann