hneemann/Digital #432: Custom circuits don't adjust inputs to height.

bcrew1375, 2020-03-31:
I'm not sure if this is an intentional design choice or not, but when changing the height of a custom circuit and using it in another circuit, the inputs and the outputs stay in the same locations as the original size. Shouldn't they be proportionately distributed according to the size of the circuit?

hneemann, 2020-03-31:
This is a design choice.
The only way to change this behavior is to use a custom shape.
To do this, create a SVG template, move the pins in the template and then import the modified SVG template.

closed 2020-04-01 by hneemann

hneemann, 2020-04-01:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

bcrew1375, 2020-04-01:
Yeah, this is what I was hoping for. It works nicely for the outputs, but the inputs seem unaffected.

hneemann, 2020-04-02:
All inputs and outputs (left, right, bottom and top) are handled in the same way.

bcrew1375, 2020-04-02:
I'm not sure what's going on, but here's a screenshot of the results I got. The circuit on the left has a height of 45, the right one has a height of 50, and both have a width of 13. Both are set to the Layout shape and no other settings have been changed for either one. 
[image]

hneemann, 2020-04-02:
The pins must always match the grid.
If there is not enough space to double the distance between the pins, it will remain as it is. If there are 32 pins, the height must be at least 64, so that the distance between the pins can be increased.

bcrew1375, 2020-04-02:
Ah, okay. Yes, that stretches them as expected. Thanks very much for your work!

commit 2704270c (2020-04-01): improved the pin layout in the layout shape type, closes #432
commit 823b77e1 (2020-04-01): improved the pin layout in the layout shape type, closes #432