hneemann/Digital #433: Buttons to save/load ROM contents are grayed out

pxmonti, 2020-03-31:
I have placed a ROM component in a circuit. I can successfully edit the content with the integrated editor, but the buttons to load or save the content are grayed out.
If I need to load several values into a ROM, doing so only with the embedded editor can be really tedious and error prone.
How can I load/save the contents of ROMs from files?

hneemann, 2020-04-01:
The ROM editor has a file menu. It allows you to load logisim hex, intel hex or binary files. See #339
The buttons act as shortcuts for loading and saving as soon as a file name is available.

pxmonti, 2020-04-01:
The most obvious solution... I completely forgot about the File menu item! Thanks for the help!
Kind regards.

hneemann, 2020-04-01:
Never mind! :smiley:

closed 2020-04-01 by hneemann