hneemann/Digital #434: communication simulated circuit <--> PC serial port

paoloSantinelli, 2020-04-01:
Hi,
I am Paolo, I am a teacher and I  teach digital design in my courses. Usually, I use FPGA boards (VHDL), but now each students can't attend the Lab because of COVID19, so, I am looking for something can be used by the students at their home. Digital seems to be perfect!

I am wondering if exist a way the simulated circuit can access to the PC serial port in order to communicate with external devices.

Thank you.

hneemann, 2020-04-02:
Up to now this is not possible. (See also #354)
But how can that be useful? What could students connect to the serial port at home?
Nowadays no computer has a serial interface anymore. And if FTDI based hardware has to be connected, then a small FPGA break out board like the  [TinyFPGA-BX](https://tinyfpga.com/) can also be connected and is more usefull.

paoloSantinelli, 2020-04-02:
Yes, you are in right, when I write "serial interface", I mean a virtual COM.
The reason for my strange request is this: The Idea is that the students develop a project composed of two parts. A programmable part, based on a PIC-Board simulator, and a second part, based on a peace of hard wired logic circuit (VHDL, using Digital).  The PIC microcontroller could communicate to the hard wired logic circuit using a virtual COM.

Anyway, in the mean time I have came across to ghdlex and netpp, two library that could be useful to achieve my goal. 

https://ghdl.readthedocs.io/en/latest/examples/VHPIDIRECT.html#ghdlex-and-netpp
https://github.com/hackfin/ghdlex

It seems that using Design and GHDL simulator should be possible develop a VHDL module to communicate with COM port (or external hardware).

What is your opinion?

Thank you for your time.

hneemann, 2020-04-02:
Connecting Digital to a PIC board emulator using a pair of virtual connected COM ports?
That's an interesting idea! I have to think about it!

hneemann, 2020-04-02:
I have implemented a simple [plug-in](https://github.com/hneemann/digitalSerialPort) which allows to access the serial port.

paoloSantinelli, 2020-04-02:
That is fantastic! A huge thank!  Now I watch the plug-in.  Anyway, this is the link at the PICSimLab: https://lcgamboa.github.io/

hneemann, 2020-04-02:
Up to now there is no interrupt output, which means you have to poll the serial interface.

paoloSantinelli, 2020-04-03:
Yes it' fine! I have successfully built the plugin and loaded it into Digital. I start playing with it. I let you know

paoloSantinelli, 2020-04-03:
I played with the plug-in! It works fine!!

Thank again!

hneemann, 2020-04-04:
Please let me know if any problems arise.

closed 2020-04-04 by hneemann