hneemann/Digital #435: External component cannot run VHDL codes because of the default GHDL Options

chaowu0604, 2020-04-02:
Hi,

I have been used the Digital in my lecture for this semester, and my students like it very much! Recently, I could like to use the external process to teach how to write VHDL codes. Unfortunately, it always failed with the error messages about missing library synopsys for v08. After searching online and digging into the source codes, I believe that it might be caused by a bug. 
First, according to the following post, Altera simulation library can only be compiled in VHDL-93 mode with enabled Synopsys flavour in GHDL.
https://github.com/ghdl/ghdl/issues/39
Second, I found the ghdl options "--std=08 --ieee=synopsys" are set as default in src/main/java/de/neemann/digital/core/element/Keys.java

Could you please help to fix the ghdl option? Thanks for your great effort on this marvelous software!

hneemann, 2020-04-02:
Are you using the latest pre release version, which you can find [here](https://infdigital.dhbw-mosbach.de/)? In the latest pre release you are able to modify the options by yourself in the external components attributes.

chaowu0604, 2020-04-03:
Thanks for your prompt reply! I just tried the newest version. After removing the ghdl options, now it is working! However, it seems that I need to remove the options every time to make it work, so I am wondering if it is possible to save attributes like the settings.

hneemann, 2020-04-03:
You have to set the option every time you open the circuit? 
The options should be stored in the dig file similar to all other attributes of all components.

chaowu0604, 2020-04-04:
Sorry! I didn't realize that it was stored in the dig file since it was appeared in multiple new circuits I tried. Thanks for your help!

closed 2020-04-04 by chaowu0604