hneemann/Digital #436: How pdf is generated?

itviewer, 2020-04-03:
I try to add Chinese translation, the software GUI works fine, but Chinese fonts are displayed as ### in the generated pdf help document .
I think the correct font should be embedded in the generated pdf file, while I have no idea how to do that.

software:
[image]

pdf:
[image]

hneemann, 2020-04-03:
Ohh! There is already someone working on a Chinese translation. 
Perhaps you should coordinate to avoid double work.

hneemann, 2020-04-03:
> I think the correct font should be embedded in the generated pdf file, while I have no idea how to do that.

Me neither! The documentation is generated using a  XML->XSLT->XML-FO->PDF workflow.
I never have worked with Chinese fonts in this kind of workflow. I will try to find out.

[Here](https://github.com/hneemann/Digital/blob/master/src/test/java/de/neemann/digital/docu/DocuTest.java) is the code that generates the documentation, in case you want to look it up yourself.

itviewer, 2020-04-03:
Here is a detailed explanation and solution
https://stackoverflow.com/questions/25895984/apache-fop-displaying-with-simsun

itviewer, 2020-04-03:
> Ohh! There is already someone working on a Chinese translation.
> Perhaps you should coordinate to avoid double work.

this is my initial translation, just in case may be used.
https://github.com/itviewer/Digital

I couldn't find any discusstion about Chinese translation in the issues. Maybe someone contacted you by email and then you gave the translation file to that person?

If someone has forked the repository and is translating, please let me know.
Thanks.

hneemann, 2020-04-03:
@itviewer
I have created a branch "Chinese" which contains your first translation and which also fixes the Chinese font issue in the documentation. This branch should also work on your machine.
Maybe this branch can be used to finalize the Chinese translation.

itviewer, 2020-04-13:
@hneemann 
I have basically completed the translation #443  . Is the screenshot corresponding to the static content automatically generated? 
After completing the screenshot,I think it should be able to merge into the master branch.

hneemann, 2020-04-13:
@itviewer 
I added the screenshots and created the documentation. I have attached the document to give you the possibility to check how it looks like when it was generated on a Linux system without installed Chinese language support.

[Documentation_zh_CN.pdf](https://github.com/hneemann/Digital/files/4469879/Documentation_zh_CN.pdf)

hneemann, 2020-04-13:
I have again changed the Chinese language code from "zh_CN" to "zh".

This should enable the simulator to select Chinese as the default on systems with Chinese system language.
Could you please pull the latest changes and try something? On your system there is a file ".digital.cfg". On Linux systems the file is located in the home folder. I am not sure where it is stored on the different Windows systems. 
If you delete this file and then start Digital, Chinese should be the default language.

itviewer, 2020-04-14:
1. the pdf has no problem
2. change "zh_CN" to "zh" can indeed use the default  language for me
---
3. about language code( **not important for now**)
The language code issue may be a problem that require refactoring the multilingual support!
For internationalization(such as operating system or other software), the language code is usually defined by the form "language_country", for example:

American: en_US
Australia: en_AU
China: zh_CN
HongKong: zh_HK
TAIWAN: zh_TW

I think the difference between American English and other countries' English is relatively small, just use en is ok.
While simplified chinese language is used in China, and traditional chinese language is used in Taiwan and Hong Kong, which is quite different.

for now, I think this problem can be ignored and everything is ok.

hneemann, 2020-04-14:
If someone is found to do a translation into traditional Chinese, I will adapt the code.

I'd like to acknowledge you in the [Readme](https://github.com/hneemann/Digital#credits). How would you like me to do that?

itviewer, 2020-04-14:
> If someone is found to do a translation into traditional Chinese, I will adapt the code.
> 
> I'd like to acknowledge you in the [Readme](https://github.com/hneemann/Digital#credits). How would you like me to do that?

How about just:

> XinJun Ma(@itviewer) has provided the Chinese translation.

Thanks

hneemann, 2020-04-14:
closed with b9dd35a7bc8a596c6627c07a107c91eab2c9c99d

closed 2020-04-14 by hneemann

hneemann, 2020-08-06:
@itviewer 
I am planning a new release and some language keys have been changed or added.
Would you like to update the attached diff file?

[lang_diff_zh.xml.zip](https://github.com/hneemann/Digital/files/5033124/lang_diff_zh.xml.zip)

itviewer, 2020-08-07:
> @itviewer 
> 
> I am planning a new release and some language keys have been changed or added.
> 
> Would you like to update the attached diff file?
> 
> 
> 
> [lang_diff_zh.xml.zip](https://github.com/hneemann/Digital/files/5033124/lang_diff_zh.xml.zip)
> 
> 

Sorry for being late, I will finish it as soon as possible.

itviewer, 2020-08-10:
@hneemann 
updated
[lang_diff_zh.zip](https://github.com/hneemann/Digital/files/5048903/lang_diff_zh.zip)

hneemann, 2021-01-15:
@itviewer 
Someone did a translation for traditional Chinese. Therefore, following your advice, I had to implement a solution that can handle different variants of the same language.
Would you try build [v0.25-271-g66de848](https://infdigital.dhbw-mosbach.de/) and confirm that the correct Chinese language variant is still selected if the .digital.cfg file is deleted? I don't know exactly where the file is located on Windows systems. On Linux the file is in your home folder.

itviewer, 2021-01-16:
@hneemann 
tested it, no problem.