hneemann/Digital #438: Quick question about input/output ports

shawty, 2020-04-09:
I'm trying to construct a small 8 bit r/w register using the following circuit:

[image]

It's operation is very simple, if the R/W line is 0 when the clock goes high, then the data on the 8 bit data line is written to the memory, if the R/W line is 1 when the clock goes high, the data from the memory appears on the data line.

As the help for this (and for that matter the output pin also.. I've tried them both) states, when the pin is used to connect this circuit as an embedded circuit in a larger circuit, the pin is bi-directional:

[image]

When ever I use this, or the output pin in such a fashion, I can never get it to work, and get an error every time about there being more than one output active at a time.

How can I use these IO pins as a bi directional data port for my component, I don't want to have to put both an In and an Out on the component unless there is no other way, as this is designed to be a register in a larger circuit design.

hneemann, 2020-04-10:
Different things are going on there:
1. To be precise, there are no bidirectional inputs and outputs. Inputs and outputs are replaced by a direct wire connection and are simply no longer there when a circuit is used as a embedded circuit. And the wires are bidirectional.
2. Only in the top level circuit, the input shape outputs a value that is similar to any other logic gate. This means that if the input outputs a value and the RAM also outputs a value, this generates a short-circuit error message. To avoid this, you can configure the input such that it can be in a high-z state.
3. The RAM element outputs a value if ld is high, regardless of the clock. I'm not sure your wiring of ld and str makes sense. It depends on the intended use.

shawty, 2020-04-10:
Thanks :-)

So to be clear, what I'm trying to do is have a register, that has one 8 bit data port, One 1 bit Read not Write port and One 1 bit clock port.

The idea is this
IF R/W = 1 then the register is in "Read Mode" and what ever 8 bit value it is holding appears on the data Port when the clock goes high.
IF R/W = 0 then the register is in "Write Mode" and what ever 8 bit value is present on the data port is stored in the register when the clock goes high.

If there is a better way of doing what I'm attempting, then I'm all ears :-)

If not, then I'll probably end up breaking out the java compiler and writing a component from scratch.

Cheers
Shawty

hneemann, 2020-04-10:
This circuit does the trick: :smiley: 
[image]

shawty, 2020-04-11:
AHHH Indeed it does.  I did try something similar myself previously, it was the addition of the driver on the data line that fixes it, that's what my first version didn't have.

/me "slaps self...."

Thanks.

closed 2020-04-11 by shawty