hneemann/Digital #444: A java exception occured

suluoniu, 2020-04-18:
I can run the code in eclipse,it works well. but when i export as runnable jar file, it  can't work,"A java exception occured“
win10 64
java 1.8.0-251
java&javac can work well in cmd window

hneemann, 2020-04-18:
Can you please attach the console output that will be output when the command line

`java -jar Digital.jar`

is being executed.

suluoniu, 2020-04-18:
Exception in thread "main" java.lang.ExceptionInInitializerError
        at de.neemann.digital.lang.Lang.get(Lang.java:36)
        at de.neemann.digital.gui.Main.<clinit>(Main.java:106)
Caused by: com.thoughtworks.xstream.io.StreamException:
        at com.thoughtworks.xstream.io.xml.StaxDriver.createReader(StaxDriver.java:102)
        at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1094)
        at de.neemann.gui.language.Bundle.<init>(Bundle.java:40)
        at de.neemann.digital.lang.Lang.<init>(Lang.java:89)
        at de.neemann.digital.lang.Lang.<init>(Lang.java:88)
        at de.neemann.digital.lang.Lang$InstanceHolder.<clinit>(Lang.java:25)
        ... 2 more
Caused by: javax.xml.stream.XMLStreamException: java.net.MalformedURLException
        at com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl.setInputSource(Unknown Source)
        at com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl.<init>(Unknown Source)
        at com.sun.xml.internal.stream.XMLInputFactoryImpl.getXMLStreamReaderImpl(Unknown Source)
        at com.sun.xml.internal.stream.XMLInputFactoryImpl.createXMLStreamReader(Unknown Source)
        at com.thoughtworks.xstream.io.xml.StaxDriver.createParser(StaxDriver.java:230)
        at com.thoughtworks.xstream.io.xml.StaxDriver.createReader(StaxDriver.java:100)
        ... 7 more
Caused by: java.net.MalformedURLException
        at java.net.URL.<init>(Unknown Source)
        at java.net.URL.<init>(Unknown Source)
        at java.net.URL.<init>(Unknown Source)
        at com.sun.org.apache.xerces.internal.impl.XMLEntityManager.setupCurrentEntity(Unknown Source)
        at com.sun.org.apache.xerces.internal.impl.XMLEntityManager.startEntity(Unknown Source)
        at com.sun.org.apache.xerces.internal.impl.XMLEntityManager.startDocumentEntity(Unknown Source)
        at com.sun.org.apache.xerces.internal.impl.XMLDocumentScannerImpl.setInputSource(Unknown Source)
        ... 13 more
Caused by: java.lang.NullPointerException
        ... 20 more

hneemann, 2020-04-18:
I assume that something went wrong when creating the jar file. It seems to me that resource files are missing in the jar. Could you please install maven and then run 

`mvn install`

from a comand line to build the jar file?

hneemann, 2020-04-18:
By the way: If you are interested in the latest changes and do not want to build your own modifications, you can simply use the latest [pre-release](https://infdigital.dhbw-mosbach.de/).

suluoniu, 2020-04-18:
sorry, it works well in cmd line. Maybe Eclipse have some bug.

suluoniu, 2020-04-18:
[ERROR] Failures:
[ERROR]   TestSyntax.testSyntax:39 expected:<0> but was:<1>
[INFO]
[ERROR] Tests run: 730, Failures: 1, Errors: 0, Skipped: 0
[INFO]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  02:25 min
[INFO] Finished at: 2020-04-18T19:21:00+08:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M1:test (default-test) on project digital: There are test failures.
[ERROR]

I redownload and recompile it in cmd , it's failure

hneemann, 2020-04-18:
Did you change any of the language files?
This error means that you have modified a german or a english language key, that they now don't use the same number of parameters.

suluoniu, 2020-04-18:
yes, i del some language file.

closed 2020-04-18 by suluoniu