hneemann/Digital #445: cannot start vivado

suluoniu, 2020-04-19:
I set the toolchian to the Basys3.config.  but when i click the menu basys3-export vhdl&start vivado
it remind me : cannot run program "vivado",createProcess error=2.

win10
vivado2015

hneemann, 2020-04-19:
The Vivado binary is not found. I suspect that Vivado also does not start when `vivado` is entered on the command line.
You can either correct the search path on your system or set the correct path in the file `BASYS3.config`:

``` XML
<toolchain name="BASYS3" clockGenerator="clockGenerator">
    <commands>
        <command name="Export VHDL" requires="vhdl" />
        <command name="Export VHDL &amp; Start Vivado" requires="vhdl" filter="true" timeout="0">
            <arg>[SET FULL PATH HERE]/vivado</arg>
            <arg>vivado/{?=shortname?}.xpr</arg>
        </command>
        <command name="Export Verilog" requires="verilog" />
        <command name="Export Verilog &amp; Start Vivado" requires="verilog" filter="true" timeout="0">
            <arg>[SET FULL PATH HERE]/vivado</arg>
            <arg>vivado/{?=shortname?}.xpr</arg>
        </command>
    </commands>
    ...
```

suluoniu, 2020-04-19:
1, i add the vivado path to the System Environment variable，   i can run vivado in cmd line. but the menu  does still not work.
2, i add the full path C:\Xilinx\Vivado\2015.4\bin\vivado.bat ,it worked.

suluoniu, 2020-04-19:
1, i add the vivado path to the System Environment variable， i can run vivado in cmd line. 
<command name="Export VHDL &amp; Start Vivado" requires="vhdl" filter="true" timeout="0">
            <arg>vivado.bat</arg>
            <arg>vivado/{?=shortname?}.xpr</arg>
        </command>

it worked

closed 2020-04-19 by suluoniu