hneemann/Digital #447: i have some questions.

dbwodlf3, 2020-04-20:
hello.

i am interesting in this project.

i used Logisim and i found this project.

i would like to contribute this project. then, if there are some docs for architecture about codes could i get docs?

i don't know where are docs about codes.

just i have to analysis from source codes?

hneemann, 2020-04-20:
Currently there is no detailed documentation of the source code. However, all packages contain a java-doc file (package-info.java), which describes what the package does and how it does it. But even this description is not as detailed as it should be.
If the documentation is extended, I will extend this kind of documentation. I am not a friend of documentation outside of the source code, like Word files, wikis or similar.

dbwodlf3, 2020-04-20:
thank you.

i try to read java-doc files and more study.. and get some information.

and come back!

closed 2020-05-20 by hneemann