hneemann/Digital #450: VHDL, how to use external custom library or packages

paoloSantinelli, 2020-04-27:
Hello everybody

I would like to use vhdl packages in order to simplify the code inside an external component. How could I do it?  I tried to build the package using the command  ghdl -a my_package.vhd, move the obtained file work-obj93.cf in to an appropriate dir (/my_package_dir),  and then, I added  -P/my_package_dir at the options of the external component. It doesn't work, when I try to check the external component, I get the message: 'unit "elementary_gates_" not found in library work"....

Any suggestion will be appreciated

Thank you very much
Paolo

hneemann, 2020-04-28:
Do you know how this works on the command line using GHDL without Digital?

paoloSantinelli, 2020-04-28:
Hi Helmut,

I didn't try to do it in GHDL, even if theoretically I know how to do it
using command line in GHDL. So far I have only built my package using ghdl.

So, your advice is to try in advance to do all the work in  cli ghdl, and
then try to move into Digital ?

Best.

Paolo

hneemann, 2020-04-28:
If you know how to set the command line options in GHDL to make your workflow work, I can try to figure out how to set the same command line options in Digital.

paoloSantinelli, 2020-04-28:
Ok, Helmut,

I try to the work in cli GHDL and then I let you know .

Thank you again.

Paolo

paoloSantinelli, 2020-04-28:
Hi,

here there are the stages to build the project by cli GHDL:

cd my_working_dir
$ ghdl -a mypackage.vhd
$ ghdl -a comb_net_struct.vhd
$ ghdl -e comb_function

No error messages are produced by the previous stages.

all the files are in the current directory.

- mypackage.vhd contains  a package with a couple of components;
- comb_net_struct.vhd contains a simple structural network and uses the
package.
- comb_function is the top entity.

Thank you.

Paolo

paoloSantinelli, 2020-04-28:
Sorry, 
here there are my files
[files.zip](https://github.com/hneemann/Digital/files/4545539/files.zip)

paoloSantinelli, 2020-05-02:
I Have solved.

My problem was not related with Digital.

To build a vhdl library to use with Digital in external module:

1. Create a file with packages, entity-architectures and all you want to include;
2. Build a library using ghdl:

    ghdl -a --std=08 --work=my_lib my_lib.vhd

   as result you should get the file my_lib-obj08.cf

3. To use the package and the entities inside the library (package name "elementary_gates"), in the source code, in the Basic tab inside the component External:
...
   library my_lib;
   use my_lib.elementary_gates.all;
...
4) Insert the path to the library In the "Options" tab inside the component External:

   --std=08 --ieee=synopsys -P/path_to_the_library_directory

Thank you and sorry for the time i have stolen

hneemann, 2020-05-02:
Never mind! :smiley:

closed 2020-05-02 by hneemann