hneemann/Digital #452: Can you add adurino components to digital?

liukenny961, 2020-05-18:
I am a digital user from China. I want to use digital to build a large circuit, but adurino neno is needed. Can you add adurino components such as adurino neno to digital? In this way, l can use adurino neno to write programs to EEPROM. In addition, l hope adurino neno's programming interface can use C++ code instead of modular programming like scratch. Or both can be added at the same time. Thank you!

In addition: my English is not very good, and there may be some grammatical mistakes. I hope you can understand my words! If there is something I have not say clear, you can ask me a question and I will answer it as soon as possible.
[image]
[image]

liukenny961, 2020-05-18:
So what is the way to do this in digital? I'm afraid not.

hneemann, 2020-05-18:
That's right! There is currently no AVR core that can be used as a digital component, and I don't think there will be one anytime soon.

liukenny961, 2020-05-18:
Will digital add this feature in the future?

liukenny961, 2020-05-18:
l think it's a difficult work.

liukenny961, 2020-05-18:
OK.

closed 2020-05-19 by hneemann