hneemann/Digital #453: Is there a way to define animated/interactive custom component shapes?

Tompouce-onGit, 2020-05-18:
In the components library there are a few components whose appearence change depending on certain factors. For instance the segments of displays and LEDs change color change depending on the value at their inputs, buttons change their shape between pressed and released when clicked, and so on.

I was wondering if there is a way to have such interaction features in *custom* components because, earlier today, I was editing a component shape in Inkscape and noticed the "interactivity" panel in the object properties, so I figured it would at least be possible to have mouse-related interactions of some sort (if only I knew more about how to use Inkscape).

Although not essential, I think it would be rather useful to convey information about a component's state, that would otherwise require the addition of LEDs or displays, on the component itself.

For example I was editing the shape for a bi-directional tri-state buffer when I noticed the aforementioned Inkscape feature

[image]

and I thought it would be nice to be able to visualy indicate in which direction the component was transmiting data, by changing the fill color of one arrow or the other.
I can also think of things like displaying the value held by a register on the register itself, or the input selected by a plexer/encoder/decoder, which could be quite useful for debugging. Things like that.

Is it possible to do currently?
If not, do you think it would be worthwhle to implement?
As I said I don't think it's strictly essential, and since the current system for custom shapes just imports an SVG file and maps the inputs and outputs to the circles with the right ID I don't know if it could support that, so if it represents to much work I'd understand if you wouldn't want to do it.

hneemann, 2020-05-18:
Adding such a state dependent shape is only possible if the component is implemented in Java. (To allow such flexibility, a full-fledged programming language is required anyway.)
In this case, you are completely free in the implementation of the shape.
This [repository](https://github.com/hneemann/digitalCustomComponents) contains an example that shows how to do this.

Tompouce-onGit, 2020-05-18:
I see, well in that case I'll have to learn a bit of Java but I think I can handle that.

closed 2020-05-19 by hneemann