hneemann/Digital #454: Is 74154 symbol wrong?

eliask05, 2020-05-18:
According to the datasheet, the outputs (Y0 - Y15) and control inputs (G1 & G2) should all be active low. In the symbol they are labelled active high.

Thanks!

liukenny961, 2020-05-19:
Is it? I also want to check it out.

eliask05, 2020-05-19:
Clarification: the device works as intended, i.e. outputs and control inputs are active low but the symbol does not reflect this.

hneemann, 2020-05-19:
It seems to me that the naming of the inputs and outputs depends on the manufacturer. The available data sheets do not seem to be consistent in this respect.

eliask05, 2020-05-19:
Understood and agreed. I am OCD and I like active low pin names to have an overbar but no big deal. I just modified the SVG to my liking and I am using it. Closing the issue.

closed 2020-05-19 by eliask05

hneemann, 2020-05-19:
@eliask05 
It is not necessary to create a custom SVG shape. 
Simply rename the inputs and outputs like this: "~G0" or "~Y0"

eliask05, 2020-05-19:
Doh! Thanks.