hneemann/Digital #458: Question: TeX math support

eliask05, 2020-05-20:
Is there a list of what TeX math constructs are supported in text boxes? subscripts and superscripts are obvious, but is there anything else? \sum and \prod would be nice for SOP and POS forms.

And, while we are at it, would direct TikZ export be too much to ask? :relaxed:

hneemann, 2020-05-20:
> Is there a list of what TeX math constructs are supported in text boxes? subscripts and superscripts are obvious, but is there anything else? \sum and \prod would be nice for SOP and POS forms.

There is the ~ character which creates a overbar. Digital is a digital circuit simulator and not a type setting system! :smile: 

> And, while we are at it, would direct TikZ export be too much to ask? 

I do not like TikZ very much, I am a fan of the Inkscape LaTeX export. Please have a look at [this](https://www.ctan.org/tex-archive/info/svg-inkscape). It allows to include SVG graphics and replace the text contained in the SVG by LaTeX commands. The result is a included SVG file with the text set by LaTeX. This looks really nice, and it works seamlessly on the fly.

eliask05, 2020-05-20:
I get that including a full-blown (La)TeX renderer might be a bit too much but a few commonly used symbols would be welcome (\sum and \prod, for example). For now, I am just using the Character Map application in windows to simply paste special characters in the text box. Playing around, I found that surrounding text with $ makes it italic. For example:
"Ex. 4.30
Realization of $f(x_1,x_2,x_3)=$∑$m$(0,2,3,5)
with a 74151A."
This results in the attached image. Is this intentional? And if so, any other similar tricks?
[image]

hneemann, 2020-05-22:
Yes! The '$' activates the math mode, which in case of Digital only means that a italic font is selected. But I would use it like this: $f(x_1,x_2,x_3)=∑m(0,2,3,5)$

closed 2020-05-25 by eliask05