hneemann/Digital #466: Seven-segment display flicker clarification

xotmatrix, 2020-05-28:
I had to examine the source code before I understood how the "Avoid Flicker" persistence control of the seven-segment display worked. In my circuit it didn't appear to do anything.

I think the documentation could be improved by saying the display persists while the common line is in a High-Z state. 

Also note the help presumes a common cathode display but the persistence control works for common anode as well.

My common cathode circuit de-asserted the display with a positive voltage rather than High-Z. This causes the display to clear/flicker and raises the question: should the persistence control work for any non-asserting signal, not just High-Z?

hneemann, 2020-05-28:
If one of the LED pins is in high-z state, the LED keeps the state it had before. This is to be able to operate a multiplexed display flicker free.
If I would also freeze the state when, for example, both pins are set to LOW or HIGH, the display could not be switched off.

xotmatrix, 2020-05-28:
Ah, that does make sense. My circuit is a multiplexed 4-digit hex display. The LEDs are driven by a modulo-4 selector built from a 7400 Quad 2-input NAND. I replaced the 7400 with a 7403 (open drain with High-Z output) and fixed the flicker. It may be a better choice my physical implementation as well, though I have none on hand.

commit b7594977 (2020-05-28): Improves the documentation of the seven segment display; closes #466
closed 2020-05-28 by hneemann