hneemann/Digital #468: IEEE 754 components

tux2603, 2020-05-28:
One thing that I've found missing for more complicated builds are built in components for IEEE 754 arithmetic operations. It's possible to implement them using existing components, but that takes time and is somewhat error prone. Since VHDL has built in operations for floating point arithmetic, would it be possible to add another set of components to the arithmetic menu for IEEE 754?

hneemann, 2020-05-29:
It's easy to implement IEEE 754 operations as Digital builtin components. But what about HDL export?
I have never tried to use floating-point arithmetic in an FPGA. Is such code synthesizable? Can you just use the asterisk and the VHDL compiler utilizes floating point hardware? That would be hard to believe but maybe it is possible nowadays?

tux2603, 2020-05-29:
There's a library/module called float_pkg that you import, and then you can just define your vectors as floating points. As far as I've seen, as long as your board supports >= VHDL 2008, it will be synthesizable.