hneemann/Digital #469: Please restore ability to use narrow shapes by default

Parakleta, 2020-05-29:
Commit 9e2aab9a90 removed the option to control wide shapes, and at the same time made IEEE shapes wide by default. Can we please have the option to choose back, or make non-wide the default again.

Thanks.

hneemann, 2020-05-29:
What is the use case? 
If you do not want to select the small shape over and over again, simply copy a gate that already uses a small shape.

Parakleta, 2020-05-30:
I figured out to copy paste gates but that feels clumsy. The main use is that the NOT gates fit between wires when they’re the smaller shape, and the smaller shapes are what are used by the autogeneration algorithm.

hneemann, 2020-05-30:
So you would like to use the IEEE wide shapes for the "normal" gates but the small shape for the inverter?

Parakleta, 2020-05-31:
I personally prefer the small gates in all instances (they look closer to the shape in printed material I have), but it’s the NOT gates that make a material difference.

hneemann, 2020-05-31:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) build.

closed 2020-05-31 by hneemann

Parakleta, 2020-06-03:
Thanks, that’s fine. Can I ask why you keep that whole `try..catch` block that defaults to wide shapes at all?

hneemann, 2020-06-03:
The `getDescription()` method is also able to load the description of a custom component, which may require loading the corresponding `*.dig` file. For this reason it can throw an IOException. But if the user inserts such a component into the circuit, the description must already be loaded. Therefore the error cannot occur in this case. And therefore the catch block can be empty.

commit 21d5ab6a (2020-05-31): uses the small shape for the NOT gate also in IEEE shape mode; closes #469