hneemann/Digital #470: Fehlermeldungen

guidogratzke1, 2020-05-29:
version: v0.24 

-- Fehler beim erzeugen der Schaltung - verwendete Bauteile aus dem Programm ;
-- Fehler beim simulieren - Schaltung scheint zu oszillieren . Habe dann die Schaltung auf eine Steckplatine aufgebaut und vermessen bis 7 Mhz keine Oszillation obwohl die gewünschte Frequenz nur 1 Mhz sein soll. Auch ein Design der Schaltung in einem FPGA zeigte keine Oszillation in der Timinganalyse. Mein verwendetes OS ist Windows 10 Pro 64bit Version 1909.
[image] 
[image]

hneemann, 2020-05-29:
Können Sie die Schaltung als ZIP an dieses Ticket anhängen?

guidogratzke1, 2020-05-30:
Ja, hier ist nun die gewünschte zip Datei mit den Schaltungen.
[Schaltungen.zip](https://github.com/hneemann/Digital/files/4705123/Schaltungen.zip)

hneemann, 2020-05-30:
Die Oszillation konnte ich nicht reproduzieren! Haben Sie evtl. noch eine Schaltung, welche die Reproduktion der Oszillation vereinfacht?

Eine Version mit dem behobenem Nullpointer-Fehler finden Sie [hier](https://infdigital.dhbw-mosbach.de/).

guidogratzke1, 2020-05-30:
Anbei sende ich eine Zip Datei mit einem Screenshot und zwei Dateien piposrg4.dig und mul8.dig.
Die Datei piposrg4.dig bildet die Grundlage für die Datei mul8.dig. Der Screenshot zeigt den Fehler.Wenn man mehrere Durchläufe der Simulation laufen lässt so tritt der Fehler auf.
[Fehler.zip](https://github.com/hneemann/Digital/files/4705557/Fehler.zip)

hneemann, 2020-05-30:
Kann es sein, dass die ZIP Datei keine Dateien enthält?

guidogratzke1, 2020-05-30:
Anbei nun aber die richtige zip Datei hatte mich vertan und bitte daher um Entschuldigung .
[Anmerkung 2020-05-30 130533.zip](https://github.com/hneemann/Digital/files/4705688/Anmerkung.2020-05-30.130533.zip)

hneemann, 2020-05-30:
Wenn SIe die Schaltung im Gatterschrittmodus starten, sehen Sie, dass die Oszillation durch die asynchronen Set und Clear Eingänge der FlipFlops verursacht wird.
Es handelt sich meines Erachtens um eine Schwäche der Schaltung und nicht um einen Fehler im Simulator.

guidogratzke1, 2020-05-31:
Ich konnte nun den Fehler beheben in dem ich das JK-FF asynchron mit dem D-FF asynchron ausgetauscht habe.Was ich zum erstellen eines Registers gelernt habe ist die Tatsche das man darauf achten soll nur eine Gatter Art zu verwenden (idealer Weise Nand) so wird die Simulation einfacher.
Das lassen sich Latenzen besser erkennen. Danke für die Mühe und Geduld von ihnen mir bei meinem Problem zu helfen.

hneemann, 2020-05-31:
Nichts zu danken! :smile:

closed 2020-05-31 by hneemann