hneemann/Digital #471: RAM appearance/manual data write

edson-acordi, 2020-06-04:
First, I would like to congrats you for the great software!
I have been used logisim evolution, but now, I found the Digital software.
I would like to suggest modifying the RAM available in Digital to work and have an option to change the appearance as in logisim (showing the data) and the option to write manually the data with built-in hex editor (like poke tool in logisim ).

hneemann, 2020-06-04:
If the simulation is running you can click on the RAM to open a hexeditor which allows to modify the RAMs content.
It the RAM is somewere in a sub circuit, you can open the "Measurement Table" (F6) which allows to access all available RAM components used in the running circuit.

edson-acordi, 2020-06-04:
Hi hneemann, thank you for your reply!
Cool, I understood!
Is there a way to save to a file the content of the EEPROM and ROM? Moreover, could this feature implemented in RAM?
I have done some tests and I could not find a way to do it.

hneemann, 2020-06-04:
At the ROM and the EEPROM the hex editor has a file menu which also offers a save entry.

edson-acordi, 2020-06-04:
Hi, this menu is not shown.
Look below if I am doing right!
Thank you.

[image]

[image]

hneemann, 2020-06-04:
The RAM has no file menu, but the ROM/EEPROMs have. Press the "Edit" button, and the hex editor shows up with a File menu. See #433, #339

If you want to store what is written to a RAM component, replace the RAM with a EEPROM.

edson-acordi, 2020-06-04:
Ok, now I saw it!
Thank you!

closed 2020-06-04 by hneemann