hneemann/Digital #472: Running Statistics from command line

moita, 2020-06-05:
Like Running Test case from command line, i think getting statistics from command line can improve the automatic batch evaluation  for student tests.
Is that possible?

hneemann, 2020-06-05:
What kind of statistics are you thinking of?
And what should the output look like?

moita, 2020-06-05:
The same table (text format) as in Simulation> Circuit statistics.
For example:
In an exercise, I propose to students to change FF-JK to FF-D.
They can send the same circuit and the results will be good. With the statistics, I can verify that they are really using D.

hneemann, 2020-06-05:
But you have to write a simple parser to analyze the output generated by Digital.
And in this case it might not be much more complex to analyze the dig file directly.

moita, 2020-06-05:
In addition, I am running the circuit check inside Moodle.

hneemann, 2020-06-05:
> It's true but it gets simpler in a small table, and that job is already done. 

I don't know what programming language you use, but I'm sure there is a XPath library available. And that makes it a one-liner. :smile:

moita, 2020-06-05:
You're right. You are doing an excellent job. Thanks

closed 2020-06-05 by moita

hneemann, 2020-06-11:
@moita 
in the latest [pre-release](https://infdigital.dhbw-mosbach.de/) i have implemented a new command line interface.
This new command line interface also allows to optain the circuit stats.