hneemann/Digital #473: Contribution: SVG Export from the command line

sedwards-lab, 2020-06-08:
I wanted the ability to convert a whole bunch of .dig files (homework from my students) into .svg files to make them easier to preview as PDF. I couldn't find that feature, so I added something like it.

I followed the pattern for your (very useful) CommandLineTester class and managed to get it to work.

It's just a single source file that can be dropped into src/main/java/de/neemann/digital/draw/graphics/ and invoked with something like

java -cp ./target/Digital.jar de.neemann.digital.draw.graphics.CommandLineSVGExporter foo.dig foo.svg

It's primitive and I don't know what your policy is about test cases, whether this is the right package in which to put something like this, or whether I'm even calling it the way you intended, hence I'll skip the push request.

Regardless, thanks for a very impressive piece of software. I teach an introductory digital design class and was sad to see Logisim succumb to obsolescence. Digital is a worthy successor!

[CLE.zip](https://github.com/hneemann/Digital/files/4744212/CLE.zip)

hneemann, 2020-06-08:
Thank you for your contribution! This is a good idea that should be implemented.
But I think that with this enhancement it will be necessary to provide Digital with a reasonable CLI interface that can be extended more easily in the future.
I will think about it and then implement your idea.

sedwards-lab, 2020-06-08:
Agreed. A proper CLI is the way it should ultimately be implemented. Thanks
again for all your work.

hneemann, 2020-07-09:
The new CLI also replaces the old testing command line discussed in #353.