hneemann/Digital #474: Test vector, sync first values

moita, 2020-06-08:
Dear Neemann,
I'm asking students to design an 4 bit assyncronous cascaded counter, up or down, decade or modulo N , all this random values. Design the circuit to set the initial value is out of context now, and this kind of counters allways gives diferent initial condition.
There any way to solve this? Or is possible to create a test file, that sets the clock until sync a first value,  and start the sequential test after that?

moita, 2020-06-08:
[DecadeJK.zip](https://github.com/hneemann/Digital/files/4746249/DecadeJK.zip)
Take as example this decade counter. It never starts 0000

hneemann, 2020-06-08:
This is not yet possible and also very difficult to implement, as there is no concept for a conditional test execution so far.
What would be possible is to give the test access to the current state of the circuit, which would allow to make the test dependent on the state of the circuit.

closed 2020-06-08 by hneemann

hneemann, 2020-06-08:
This problem was much easier to solve than I thought at the beginning. :smile: 
Please download the latest [pre-release](https://infdigital.dhbw-mosbach.de/) and try the attached circuit.

[DecadeJK_new.zip](https://github.com/hneemann/Digital/files/4748302/DecadeJK_new.zip)

moita, 2020-06-08:
You are TOP!
Don't forget to have a security for the while cycle. Could have situations where the FALSE conditions couldn't be reached.
I have another problem. While I was trying to teach my students how multiplexed  seven segment displays work, they just flicker at 200hz. Is there a solution?
I send you a simple example. Tell me if I'm being boring please.
[MuxDisplay.zip](https://github.com/hneemann/Digital/files/4748645/MuxDisplay.zip)

xotmatrix, 2020-06-08:
@moita See #466

moita, 2020-06-09:
Right! Tanks

commit e902d3e9 (2020-06-08): allows the test case to react on the model state, closes #474