hneemann/Digital #475: Analysis tool only partially supports format setting

xotmatrix, 2020-06-09:
The analysis tool only partially supports the format setting. It uses the desired operators but appears to always uses overbar for complement.

[image]

It would also be very helpful if this analysis could be saved or copied to the clipboard.

Affects v0.24, Windows 10

hneemann, 2020-06-09:
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/).

closed 2020-06-09 by hneemann

xotmatrix, 2020-06-09:
It took me a moment to figure out the change was in the File menu. It seems to work perfectly. Thank you very much for making this tool an even greater pleasure to use.

commit 49e5e397 (2020-06-09): adds a plain text export in the table dialog, closes #475