hneemann/Digital #476: Text formatting

nortont, 2020-06-12:
I have not been able to locate documentation of text formatting.
I am trying to put an underscore between 2 words but the second word becomes subscript.
I tried the LATEX method of \_ but that did not work

In another issue I have been able to put a inverse line above text using ~ but need to put the next word adjacent with no space and cannot see how to do that.

closed 2020-06-12 by hneemann

hneemann, 2020-06-12:
Please try the letest [pre-release](https://infdigital.dhbw-mosbach.de/). Now "\\_" to escape a underscore is working.

> In another issue I have been able to put a inverse line above text using ~ but need to put the next word adjacent with no space and cannot see how to do that.

try ~{A}B

nortont, 2020-06-13:
Thank you - that worked perfectly.
Is there any further documentation on these commands?

This is truly a great software but may I suggest a name change as telling my colleagues to install Digital without giving them a link to the Git page means they will never find it.
Maybe something like "Digital Logic Design" or "Digital Logic Builder".

Regards,

Tim Norton

PH: 0437 169 195

FAN: NORT0144

ID: 2185104

hneemann, 2020-06-13:
> Is there any further documentation on these commands?

The examples/misc folder contains an example that illustrates text formatting.

commit 2e74b010 (2020-06-12): reworked text formatting, closes #476