hneemann/Digital #479: test signal not found

moita, 2020-06-21:
When you use don't care, you can get alternative solutions for the same problem.
In some cases your simplified solutions has different variables.
If you ask students do design the simplified circuit and use the same test case for command line evaluation it don't work because you get 
    Error running tests: because one input or more isn't there.
There are any chance to  automatically consider  as don´t care  the not used variables ?

hneemann, 2020-06-21:
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/). There is a new command line option.

moita, 2020-06-21:
I tried with command:
java -cp Digital.jar CLI test -circ NAND.dig -tests teste1.dig
Teste1.dig has a W X Y Z test table.
NAND.dig is a simplifies function that has only W X Z inputs.
I got the following error message. (Y not found!)
Use default settings!tv: Sinal de teste Y não encontrado no circuito!Tests have failed.

moita, 2020-06-21:
You said that there are a new command line option. Where can I find it?

hneemann, 2020-06-22:
if you type 
```
java -cp Digital.jar CLI
```
a help text is shown. The documentation also mentions the new option `-allowMissingInputs`.

closed 2020-06-22 by hneemann

moita, 2020-06-22:
Working well, thank you!
I have two other suggestions:
1- I think that selecting a wire with the right mouse button would be very practical.
2-I'm colorblind. There are many students who are too. It may be strange for you, but the HIGH, LOW colors you use are very confusing to me. Is there any chance of being able to specify these colors or simply print 0/1 inside the input / output circles?

hneemann, 2020-06-22:
> I think that selecting a wire with the right mouse button would be very practical.

It is already possible to select a wire with CTRL down.

> I'm colorblind. [...] It may be strange for you, but the HIGH, LOW colors you use are very confusing to me.

I have always resisted implementing different color schemes because they do not fit well into the architecture and it seemed unnecessary. 
However, supporting color-blind users is a really strong argument. I'll work something out.

hneemann, 2020-06-22:
@moita 
I am thinking about adding support of different color schemes. But I find it very difficult to select colors that are suitable for color blind people.
Would you help me with that? If you can handle building the simulator, you could directly change the colors in the code. I would like to save myself the trouble of making the colors selectable via the user interface.

moita, 2020-06-22:
There are different types of color-blind people. I hardly distinguish between the two levels of green (I had to ask for the colors :-)) that you use. For me it's better if you chose a darker green and a bright yellow. 
Let me know were to change in the code, ill try.

hneemann, 2020-06-22:
There is a new branch  "colorSchemes" 
The color-blind scheme is defined [here](https://github.com/hneemann/Digital/blob/colorSchemes/src/main/java/de/neemann/digital/draw/graphics/ColorMap.java).

xotmatrix, 2020-06-24:
>  I would like to save myself the trouble of making the colors selectable via the user interface.

Perhaps allowing a user to load a color scheme from a file would be a good overall enhancement.

hneemann, 2020-06-25:
@xotmatrix 
> Perhaps allowing a user to load a color scheme from a file would be a good overall enhancement.

Maybe you're right! But I think most users are happy if they don't have to worry about individual colors because suitable color schemes are available.

hneemann, 2020-06-25:
@moita 
The latest [pre-release](https://infdigital.dhbw-mosbach.de/) build supports color schemes, and one of them is aimed for color-blind people.
Please let me know if this will help you better identify the wire states.

moita, 2020-06-25:
Really better, thanks!

Atentamente,
Fernando Moita
————————————————
Departamento de Engenharia Eletrotécnica / Electrical Engineering Department
Instituto Superior de Engenharia de Coimbra / Coimbra Institute of
Engineering
Rua Pedro Nunes, 3030-199 Coimbra, Portugal

25/06/2020 à(s) 13:33:

hneemann, 2020-06-30:
@xotmatrix 
Now there is also a user definable color scheme available.

@moita 
If you feel that the scheme for colour-blind users needs to be improved, you have the opportunity to do so now. If you are satisfied, you can simply send me the ".digital.cfg" file which contains the custom scheme.

Here you can find the latest [pre-release](https://infdigital.dhbw-mosbach.de/).

moita, 2020-07-01:
Dear Hneemann,
There are different types of color blindness, and I think I have a severe
type of protanopia. For me, the colors in the configuration file work well,
but may not be the rule for others. Thanks for your consideration.
I have another suggestion.
Although you can choose the format in which you can edit a multi-bit Input
element, the value shown is always HEX. Any possibility of displaying the
input value with the same configuration defined for editing?
[image: x.png]
Atentamente,
Fernando Moita
————————————————
Departamento de Engenharia Eletrotécnica / Electrical Engineering Department
Instituto Superior de Engenharia de Coimbra / Coimbra Institute of
Engineering
Rua Pedro Nunes, 3030-199 Coimbra, Portugal

à(s) 16:26:

moita, 2020-07-01:
Consider the new cfg file with changes in the Highlighted color
Atentamente,
Fernando Moita
————————————————
Departamento de Engenharia Eletrotécnica / Electrical Engineering Department
Instituto Superior de Engenharia de Coimbra / Coimbra Institute of
Engineering
Rua Pedro Nunes, 3030-199 Coimbra, Portugal

---------- Forwarded message ---------
De: Fernando Moita <moita@isec.pt>
Date: quarta, 1/07/2020 à(s) 12:26
Subject: Re: [hneemann/Digital] test signal not found (#479)
To: hneemann/Digital <
reply@reply.github.com>

Dear Hneemann,
There are different types of color blindness, and I think I have a severe
type of protanopia. For me, the colors in the configuration file work well,
but may not be the rule for others. Thanks for your consideration.
I have another suggestion.
Although you can choose the format in which you can edit a multi-bit Input
element, the value shown is always HEX. Any possibility of displaying the
input value with the same configuration defined for editing?
[image: x.png]
Atentamente,
Fernando Moita
————————————————
Departamento de Engenharia Eletrotécnica / Electrical Engineering Department
Instituto Superior de Engenharia de Coimbra / Coimbra Institute of
Engineering
Rua Pedro Nunes, 3030-199 Coimbra, Portugal

à(s) 16:26:

hneemann, 2020-07-01:
@moita 
I didn't get those two files. Neither the screenshot nor the cfg file.

> For me, the colors in the configuration file work well, but may not be the rule for others.

You're the only person I know with this problem.That's why I can choose the colors to suit you best.
If someone with a different kind of color blindness responds, I would simply add another color scheme.

moita, 2020-07-01:
[digital.zip](https://github.com/hneemann/Digital/files/4859103/digital.zip)
[image]

hneemann, 2020-07-01:
@moita
All the wire colors have changed. Is that intentional?

moita, 2020-07-01:
yes

moita, 2020-07-01:
for me the Normal color-scheme with  Wire, wire High, Wire Low, Output and Highlighted changed as in the file cfg works well

hneemann, 2020-07-01:
The latest [pre-release](https://infdigital.dhbw-mosbach.de/) contains your color suggestions.

xotmatrix, 2020-07-03:
Thanks for all you do, @hneemann.

commit 1018e68a (2020-06-22): add an cli-option to allow missing inputs in circuits, closes #479
commit 6c8bc65d (2020-06-22): adds a new testing CLI option, closes #479