hneemann/Digital #480: Strange relay circuit behavior

DmitryVyal, 2020-06-21:
I discovered a strange behavior of a circuit containing a RelayDT group

[circuit.zip](https://github.com/hneemann/Digital/files/4808958/circuit.zip)
[image]
when I switch the input marked * to 1 everything works correctly
[image]
when I switch the input marked * back to 0 circuit state is incorrect
[image]
I don’t understand the reason, however, under certain conditions, the order of operation of the “contacts” is important for the correct operation of the circuit model

[PlainSwitchDT.java.diff.txt](https://github.com/hneemann/Digital/files/4809020/PlainSwitchDT.java.diff.txt)

hneemann, 2020-06-21:
This is an interesting one! I haven't looked at the code yet, but I think it makes a difference if there is a short moment when both switches are open or if there is a short moment when both switches are closed. There should always be a short moment where both switches are open, and there should not be the opposite.

hneemann, 2020-06-24:
This one was really hard to track down! My first guess was completely wrong!
I should never have implemented those switches! :smile: 
Thanks for pointing this out.

commit ee5fd615 (2020-06-24): fixes a bug in the switch net formation, closes #480
closed 2020-06-24 by hneemann