hneemann/Digital #483: Strange behavior of flip-flops

Lo-Dyih-Yuaan, 2020-06-23:
When connect a flip-flop's output to its own input, it seems will have an incorrect result.

[image]

hneemann, 2020-06-23:
Both circuits are working as expected. In the circuit at the top you have connected the inverted output to an inverter and then to the D-FF input. This means you are storing the value which is already stored. So nothing happens.
in the circuit at the bottom you have connected the inverted output to the kill input. This means at startup the K input is high which causes the JK-FF to toggle. After that the inverted output is low and the JK-FF stays in the "on" state.

Lo-Dyih-Yuaan, 2020-06-23:
> 两个电路均按预期工作。在顶部的电路中，已将反相输出连接到反相器，然后连接到D-FF输入。这意味着您要存储已存储的值。所以什么也没发生。
> 在底部的电路中，您已将反相输出连接到终止输入。这意味着在启动时，K输入为高电平，从而导致JK-FF切换。此后，反相输出为低电平，JK-FF保持在“开”状态。

Sorry, I misjudged whose 'Q' is the bar embellished.

closed 2020-06-23 by Lo-Dyih-Yuaan