hneemann/Digital #486: Wire junction improperly merged

xotmatrix, 2020-06-26:
When two wires are connected to a pin and form a straight line, they are merged and disconnected from the pin.

[image]

hneemann, 2020-06-26:
That' s how it works. 
To connect a pin to a wire, the wire must end at this pin. A straight wire is not connected to a pin that is placed on it.

xotmatrix, 2020-06-27:
You say "a wire" but it is clearly two wires until the software arbitrarily merges them. If the wires were colinear but diagonal the software would not "optimize" them because it does not check that case. 

The reason I bring it up is this behavior can lead to a lot of unfortunate circuit damage when playing with routing and awkward wiring to get around it. 
[image]

I'm struggling to think of a use case where merging and disconnecting is the desired result but I'll accept "won't fix" as a resolution. I understand Digital is primarily for schematic circuit representation and not physical routing. But with physical components in the parts box it is impossible to resist using it for routing since using better suited routing tools can mean a lot of wasted work while a design is being iterated upon.

hneemann, 2020-06-27:
One of the things that bothered me about Logisim was the autorouting, which never worked the way I wanted it to (which was probably my fault), and the fact that when you move parts, wires get connected to pins if they accidentally fall on a wire, which also damages the circuitry, although it rarely happens. 
I get along better with this solution. Maybe not because it's better, but because it's predictable for me.
I think in the end it's a matter of habit.

closed 2020-06-29 by hneemann