hneemann/Digital #487: Suggestions to DIGITAL.

TheInfinityUser, 2020-06-27:
This Software has so much potential. The suggestions I want to make are:-
> Making it easy to build integrated circuits, like that in logisim.
> The pins disappear when the circuit is zoomed in too much. Please fix it.
> Please add a shift register.
> Please make the wires which are more than 1-bit black, like that in logisim.
> Please allow connecting the inputs of the components unnecessary. The error really bugs me.
> Please make an unidentified wire blue, like that in logisim.

hneemann, 2020-06-27:
> Making it easy to build integrated circuits, like that in logisim.

It is super easy to create a integrated circuit: Assign a DIL shape and number the pins correctly. That's it.

> The pins disappear when the circuit is zoomed in too much. Please fix it.

Inputs and outputs do not disappear. Only the text is omitted if it becomes unreadable.

> Please make the wires which are more than 1-bit black, like that in logisim.

Why?

> Please allow connecting the inputs of the components unnecessary. The error really bugs me.

This behaviour is intended. See #36

> Please make an unidentified wire blue, like that in logisim.

What do you mean with a "unidentified wire"?

TheInfinityUser, 2020-06-27:
> Making it easy to build integrated circuits, like that in logisim.

I meant that logisim actually showed the circuit in the component tree and you could create a new circuit in the software itself and use the previous circuit as a sub circuit in the new circuit (sorry I used the word 'circuit' too many times).

> The pins disappear when the circuit is zoomed in too much. Please fix it.

I meant that when you zoom out too much for big circuits the pins become so small that they merge with the border of a component. You can actually change the colour of the input to fix the problem as you have changed the colour of the output to red.

> Please make the wires which are more than 1-bit black, like that in logisim.

Ignore that suggestion.

> Please allow connecting the inputs of the components unnecessary. The error really bugs me.

Ok... I guess I'll have to live with that.

> Please make an unidentified wire blue, like that in logisim.

Sorry, I meant "undefined", i.e., neither 0 nor more than 0.

hneemann, 2020-06-27:
> I meant that logisim actually showed the circuit in the component tree and you could create a new circuit in the software itself and use the previous circuit as a sub circuit in the new circuit (sorry I used the word 'circuit' too many times).

But that's how it works. The only difference is that a sub-circuit is stored in a separate file, which makes reuse and versioning much easier.

> I said that because I am and pretty sure most of the other users are used to logisim and it is hard to recognise the wire as a multi-bit wire.

In the settings you can enable the "Show the number of wires on a bus". Maybe that helps.

> Sorry, I meant "undefined", i.e., a wire that does not have any input value.

A undefined wire is not possible in Digital. If you try to simulate a circuit with an undefined wire, you will get an error message and the wire will turn red.

TheInfinityUser, 2020-06-27:
Are you considering on adding the shift register anytime soon? Or any other new components?

hneemann, 2020-06-27:
> Are you considering on adding the shift register anytime soon?

There are so many variants of shift registers that it is difficult to implement one that meets all needs. 
And on the other hand, a shift register is very easy to implement with the existing components. 
So in this case it is probably easier if you just build what you need.

> Or any other new components?

If it makes sense, sure.

> Also, please add an in-built intgrated circuit editor like that of logisim instead of using another software (inkscape) to draw and edit the circuit.

I think it's not worth the effort. See also #39
There are already some shape types, which cover most cases. There won't be many users that won't be satisfied with this, and they won't have a problem with Inkscape.

TheInfinityUser, 2020-06-27:
> But I think it's very tedious to use if you want a simple circuit layout.

hneemann, 2020-06-27:
You could put the shift register in a sub circuit. That'll keep the main circuit clean.

TheInfinityUser, 2020-06-27:
Ok, thanks for being active with me and replying to my emails. I'll try as hard as I can to get used to this soft

TheInfinityUser, 2020-06-27:
Also I have one more suggestion - Please add the feature to repeat a particular circuit pattern in a sub circuit. For e.g. in a demultiplexer or any other plexer, you can change the data bit count and so on,

closed 2020-06-27 by TheInfinityUser

hneemann, 2020-06-27:
This is a example of a simple generic shift register.

[shift.dig.zip](https://github.com/hneemann/Digital/files/4841060/shift.dig.zip)