hneemann/Digital #488: How does the barrel shifter work?

sudhashbahu, 2020-06-27:
I want to implement the equivalent of the C bitwise Operations D >> N and D << N, where D is 16 bit data and N is the unsigned value by which the shift is done.

I have created two barrel shifters for the two operations. The "in" input of set to the 16 bit D. The property is DataWidth=16. The "shift" property is set to bits 0-3 of N (via Splitter). 4 bits are used for the shift value  as there is no point shifting more than 16 bits.

I get the error "There are 5 bits needed, but 4 bits found. Affected are: Signal 0-3, Barrel shifter, shift"

Also, it is nor clear from the documentation how the "mode" property behaves.

hneemann, 2020-06-27:
> 4 bits are used for the shift value as there is no point shifting more than 16 bits.

You are right, but you should be able to shift 16 bits, which outputs a zero, and to do this, a 5 bits shift input is required.

> Also, it is nor clear from the documentation how the "mode" property behaves.

But the "mode" combo box seems clear to me: "Logical", "Arithmetic"  or "Rotation".
Even though "Arithmetic" and "Left" doesn't make that much sense. 
When shifting to the left, "Logical" and "Arithmetic" are identical.

sudhashbahu, 2020-06-27:
Thank you for the prompt response and clarifications.

closed 2020-06-27 by sudhashbahu