hneemann/Digital #490: Problems with labels and pin numbers.

TheInfinityUser, 2020-06-28:
When I add a pin number or assign a label to a component, it works fine when it is facing 0 degrees or 180 degrees. However, when I rotate the component 90 degrees or 270 degrees the label and/or the pin number appears sideways. please fix this issue. It also happens with input and output components. The value appears sideways when rotated 90 degrees or 270 degrees. Please fix this issue. Also, please add an option for the the position of the labels and/or pin numbers on each of the components.

hneemann, 2020-06-28:
It is intended that the labels behave in this way.

closed 2020-06-28 by hneemann

TheInfinityUser, 2020-06-28:
Then please add a way to change the direction and position of the labels.(by position I mean whether it is present on the left, right, top, bottom or centre of the component).