hneemann/Digital #492: CLI Interface?

csbrown, 2020-07-09:
Are there any options for creating a truth table file from a .dig file?  Is there any CLI interface for anything in this program at all, or is it all GUI?  If there is no CLI interface, can someone point me in a direction of where I might start implementing such a thing in a way that might be acceptable for inclusion by the repo owners?

hneemann, 2020-07-09:
A CLI interface was recently implemented and is not yet included in the latest release.
Only the last [pre-release](https://infdigital.dhbw-mosbach.de/) version already contains the CLI interface.

But up to now it is not possible to generate a truth table using the CLI. However, if you describe the use case in more detail, such a function could be added.

The CLI is implemented [here](https://github.com/hneemann/Digital/blob/master/src/main/java/de/neemann/digital/cli/Main.java).

csbrown, 2020-07-09:
My use case is that I am going to have a bunch of .dig file homework submissions from students, and it would be nice to be able to parse these into truth tables as a first pass automated grading device.

Thanks for showing me around the code.  Do you know where the functions are to create and write the various truth table formats?  I will try to go poke through the code and see what I can do.  My java is rusty anyhow, and I need to brush up for this class in the Fall.

hneemann, 2020-07-09:
You might consider using the test cases that are very well suited for the automatic testing of assignments.
See the discussion in #353.

The command line used in #353 has now been replaced by the new CLI interface, as described in the pre-release documentation.

closed 2020-07-10 by csbrown