hneemann/Digital #494: Feature Request: MQTT component

lnetwalker, 2020-07-17:
I would like to see a number of devices which just represent MQTT values. For example a output device which publishes its values to MQTT or a input device which take MQTT values and puts them in my cirquit.
With these devices it would be very easy to connect real hardware to simulated circuits.
I looked at the sources but my Java knowledge is far too basic to do it by myself. Anybody out there who is able and willing to give this idea a chance ?

hneemann, 2020-07-18:
A MQTT component is very easy to implement bacause of the [Paho](http://www.eclipse.org/paho/clients/java/) project.
There is already a [UART plugin](https://github.com/hneemann/digitalSerialPort) which can be use as a base for a MQTT integration.
If you describe in more detail what you need, it will be easy to implement.

lnetwalker, 2020-07-20:
Thanks for your attention on my suggestion. 
In the simplest case there would be Inputs and Outputs ( like Buttons and LEDs ) which are connected to MQTT topics.
Inputs represent the subscription of a topic, outputs publish their state to topics. In the first step the allowed values ( payload ) for 
the topics are 0 and 1. In a more sophisticated szenario it could be possible to have something like a regex which extracts values from a payload. 
I can support this with testing, writing documentation in German and English.

fazerlab, 2020-11-12:
With MQTT we can integrated Digital with other tools in software and hardware. I endorse and desire this enhancement.
Anyway Digital is a great tool that i used with my students . Thank you.

lnetwalker, 2021-03-02:
is there a chance that MQTT is added ? I'm willing to help whereever I can.

hneemann, 2021-03-02:
To be honest, I don't think that MQTT will be implemented as part of Digital.
This is more suitable to be implemented as a jar component in a separate project, similar to the [serial port integration](https://github.com/hneemann/digitalSerialPort).

jorge-agra, 2021-06-16:
Hi. Made a try on this, very raw and in initial stages, since I stumbled on this amazing project at the same time I was looking to MQTT.

You can check and eventually try my first attempt in: https://github.com/jorge-agra/digital-custom-component-mqtt

Just a basic MQTT send component, please exercise some forgiveness :)

Best.
JA

lnetwalker, 2021-07-08:
Hi,
until now I was busy and had no time to check it out, but it seems very promising.
Can I download a compiled jar anywhere to check it out ? ( I have currently no development Env )

jorge-agra, 2021-07-08:
Hi. Created a jar of it, and added some instructions on how to use it from Digital. Needed some tweaking to be able to load the plugin and the paho client lib.

Good luck using it, you will probably need it :)

Contact me if you need help.

Best.