hneemann/Digital #495: Unable to build circuit with 17 inputs out of NAND gates

bottle2, 2020-07-24:
I wanted to see how many NAND gates my 8 bit carry-lookahead adder would have if it was built solely using them.
The adder was created from 32 boolean expressions, later each circuit that depended on another connected with tunnels (some expressions' variables were results from other expressions). These are the 32 expressions:

```
let p0 = a0^b0;
let p1 = a1^b1;
let p2 = a2^b2;
let p3 = a3^b3;
let p4 = a4^b4;
let p5 = a5^b5;
let p6 = a6^b6;
let p7 = a7^b7;
let g0 = a0 b0;
let g1 = a1 b1;
let g2 = a2 b2;
let g3 = a3 b3;
let g4 = a4 b4;
let g5 = a5 b5;
let g6 = a6 b6;
let g7 = a7 b7;
let S0 = p0 ^ C0;
let C1 = g0 + p0 C0;
let S1 = p1 ^ C1;
let C2 = g1 + p1 g0 + p1 p0 C0;
let S2 = p2 ^ C2;
let C3 = g2 + p2 g1 + p2 p1 g0 + p2 p1 p0 C0;
let S3 = p3 ^ C3;
let C4 = g3 + p3 g2 + p3 p2 g1 + p3 p2 p1 g0 + p3 p2 p1 p0 C0;
let S4 = p4 ^ C4;
let C5 = g4 + p4 g3 + p4 p3 g2 + p4 p3 p2 g1 + p4 p3 p2 p1 g0 + p4 p3 p2 p1 p0 C0;
let S5 = p5 ^ C5;
let C6 = g5 + p5 g4 + p5 p4 g3 + p5 p4 p3 g2 + p5 p4 p3 p2 g1 + p5 p4 p3 p2 p1 g0 + p5 p4 p3 p2 p1 p0 C0;
let S6 = p6 ^ C6;
let C7 = g6 + p6 g5 + p6 p5 g4 + p6 p5 p4 g3 + p6 p5 p4 p3 g2 + p6 p5 p4 p3 p2 g1 + p6 p5 p4 p3 p2 p1 g0 + p6 p5 p4 p3 p2 p1 p0 C0;
let S7 = p7 ^ C7;
let C8 = g7 + p7 g6 + p7 p6 g5 + p7 p6 p5 g4 + p7 p6 p5 p4 g3 + p7 p6 p5 p4 p3 g2 + p7 p6 p5 p4 p3 p2 g1 + p7 p6 p5 p4 p3 p2 p1 g0 + p7 p6 p5 p4 p3 p2 p1 p0 C0
```

I cloned the repository and with Digital built v0.24-137-ga08496b (as of a08496b5b132a49093a7a93a0bf4007ef2d7687a), running it with OpenJDK Runtime Environment (build 14.0.2+12) using `java -Xms2048M -jar Digital.jar experimental` and `java -Xms2560M -jar Digital.jar experimental`, both instances were able to generate identical tables. The sha256sum of the two saved tables is `37413a273d8bec7b51d4ba36ccd3c28903d1eb9a20866683251d02f6a132777e` and `diff` found no differences either. But the tables were generated with an error in both cases:

```
de.neemann.digital.analyse.expression.ExpressionException: ExpressionListenerStore not closed
        at de.neemann.digital.gui.components.table.ExpressionListenerStore.replayTo(ExpressionListenerStore.java:62)
        at de.neemann.digital.gui.components.table.ExpressionCreator.create(ExpressionCreator.java:93)
        at de.neemann.digital.gui.components.table.TableDialog.lambda$calculateExpressions$9(TableDialog.java:723)
        at java.base/java.lang.Thread.run(Thread.java:832)
```

With the table, I tried to build a circuit solely with NAND gates, and the following error issued:

```
de.neemann.digital.analyse.expression.ExpressionException: Não há equações mínimas!
        at de.neemann.digital.gui.components.table.BuilderExpressionCreator.create(BuilderExpressionCreator.java:50)
        at de.neemann.digital.gui.components.table.TableDialog.createCircuit(TableDialog.java:642)
        at de.neemann.digital.gui.components.table.TableDialog.createCircuit(TableDialog.java:630)
        at de.neemann.digital.gui.components.table.TableDialog.access$1500(TableDialog.java:75)
        at de.neemann.digital.gui.components.table.TableDialog$26.actionPerformed(TableDialog.java:554)
        at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
        at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
        at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
        at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
        at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:369)
        at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1020)
        at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1064)
        at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297)
        at java.desktop/java.awt.Component.processMouseEvent(Component.java:6636)
        at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
        at java.desktop/java.awt.Component.processEvent(Component.java:6401)
        at java.desktop/java.awt.Container.processEvent(Container.java:2263)
        at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5012)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4844)
        at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4919)
        at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4548)
        at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4489)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
        at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2764)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4844)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

If it is needed to know anything else, I'm glad to inform.

hneemann, 2020-07-25:
I'm not sure I understood the problem.
I have tested your input to the circuit generator, and this seems to work fine. 

If you then try to create a truth table, it's is not possible because of not unique variables used.

But if you fix the internal used local variables (which should be done by the circuit generator automatically, i guess) there are only 17 inputs remaining (a0-a7, b0-b7 and c0) and you will get the truth table of an 8-bit adder which is the same for any 8-bit adder and therefore useless in respect of the carry-look-ahead logic.

hneemann, 2020-07-25:
When I played around with your equations, I was able to reproduce one of the exceptions.

Could it be that you were trying to simplify the equation for C8 which has 17 input variables? Setting the memory 2GByte is not enough for that, even with 12GByte the equation can't be simplified with QMC.

The root cause of the error messages is a Java heap space error.

bottle2, 2020-07-25:
You understood correctly the problem, but now I see how foolish on my part it was to think that whatever circuit I could build from the truth table, be it made out of two-input NAND gates or anything else, would represent the carry-lookahead circuit.

> Could it be that you were trying to simplify the equation for C8 which has 17 input variables?

Not directly, but certainly. After I generated the circuit with the equations and I fixed the internal variables, just as you did, I hit "Analysis" from the "Analysis" menu and waited until the table was generated, when the errors occurred.

I think you are right with it being lack of memory, specially for C8. Using Digital as of 98dc38540bcd649fcd2b4ffe2089d2458e841298 and with the same parameters as before, I tried "Synthesis" of the "Analysis" menu, I opened the saved table and, after a while, this appeared: "java.lang.OutOfMemoryError: Java heap space". For context, this is the full output in the terminal:

```
[AWT-EventQueue-0] INFO de.neemann.digital.gui.components.table.TableDialog - start optimization
[AWT-EventQueue-0] INFO de.neemann.digital.gui.components.table.TableDialog - optimization finished
[AWT-EventQueue-0] INFO de.neemann.digital.gui.components.table.TableDialog - start optimization
[AWT-EventQueue-0] INFO de.neemann.digital.gui.components.table.TableDialog - optimization finished
[AWT-EventQueue-0] INFO de.neemann.digital.gui.components.table.TableDialog - start optimization
[AWT-EventQueue-0] INFO de.neemann.digital.gui.components.table.TableDialog - optimization finished
de.neemann.digital.analyse.expression.ExpressionException: ExpressionListenerStore not closed
        at de.neemann.digital.gui.components.table.ExpressionListenerStore.replayTo(ExpressionListenerStore.java:62)
        at de.neemann.digital.gui.components.table.ExpressionCreator.create(ExpressionCreator.java:93)
        at de.neemann.digital.gui.components.table.TableDialog.lambda$calculateExpressions$9(TableDialog.java:723)
        at java.base/java.lang.Thread.run(Thread.java:832)
jul. 25, 2020 3:15:56 PM sun.awt.X11.XToolkit processException
WARNING: Exception on Toolkit thread
java.lang.OutOfMemoryError: Java heap space

de.neemann.digital.analyse.expression.ExpressionException: ExpressionListenerStore not closed
        at de.neemann.digital.gui.components.table.ExpressionListenerStore.replayTo(ExpressionListenerStore.java:62)
        at de.neemann.digital.gui.components.table.ExpressionCreator.create(ExpressionCreator.java:93)
        at de.neemann.digital.gui.components.table.TableDialog.lambda$calculateExpressions$9(TableDialog.java:723)
        at java.base/java.lang.Thread.run(Thread.java:832)
```

Indeed, by removed all outputs except C8 from the table through the option "Reorder/Delete Output Columns" in the "Edit" menu, the same error appears after a while: `java.lang.OutOfMemoryError: Java heap space`.

I think we may close the issue then.

hneemann, 2020-07-25:
The problem with QMC (Quine and McCluskey) is that while it always provides an exact minimum solution, the memory consumption and computing time increases exponentially with the number of input variables. Therefore QMC is not suitable for functions with many inputs. In such a case you would have to use e.g. the heuristic Espresso algorithm, which is faster and uses less memory, but does not always provide the minimum solution.
Unfortunately, I have not yet got around to implementing the Espresso algorithm.

closed 2020-07-25 by hneemann

hneemann, 2022-02-11:
see #918