hneemann/Digital #497: MUX inconsistancy

wmuece, 2020-07-28:
I have noticed that Digital's MUXes give me a consistent error message no matter how I set them up.
It goes as follows:
For 8 bit mUX: "There are 8 bits needed, but one bit is found"
For 4 bit mUX: "There are 4 bits needed, but one bit is found"
etc.
What is going with my inputs?  
Thanks.  I really like Digital... I think.
-Prof. Dean Johnson

hneemann, 2020-07-28:
There are two bit settings: First the "Number of selector bits", which is used to select the number of inputs. And second the "Data Bits" setting, which defines how many data bits are used on one of the data inputs. This way the muxer can be used to switch a complete data bus without having to add a separate muxer for each bit.

wmuece, 2020-07-28:
Helmet:  Thanks much.

Where is a good place to copy your examples if I wanted to try to use/modify them?  I noticed that copy and pasting “Adder8bit.dig"  in the “examples” folder gets me in trouble with this message when I attempt to open it:
“There are several files with identical file names, which cannot be uniquely assigned.  Affected are:
{{A list of 18 files from “examples"}}

Also, pasting that file outside “examples" gives me this message:
“The selected filename is not importable from the actual project.”

BTW & FYI:  I'm exploring the usage of your software for my fall course that has previously used Logisim in earlier semesters, with Xilinx and Basys3.   With the Covid problems we have in the US, I really now need to make a home-brew lab.  I am exploring something called a “xyLab", a feature made possible from the “Digital Design” xyBook title from xyBooks.com<http://xyBooks.com>.  Actually xyLabs really are intended for their software courses, like “Programming in C" (or C++ Java or Python or Web Programming).  However, this summer the company helped me develop a way to make the zyLab look and grade VHDL code.  The students and I just have to supply the VHDL code, such as generated by your Digital software.

- Dean Johnson
  Department of Electrical and Computer Engineering
  Western Michigan University

wmuece, 2020-07-28:
Im sorry, it is from zyBooks.com<http://zyBooks.com>, not xyBooks.com<http://xyBooks.com>.  - Dean

On Jul 28, 2020, at 9:06 AM, Dean Johnson <dean.johnson@wmich.edu<mailto:dean.johnson@wmich.edu>> wrote:

Helmet:  Thanks much.

Where is a good place to copy your examples if I wanted to try to use/modify them?  I noticed that copy and pasting “Adder8bit.dig"  in the “examples” folder gets me in trouble with this message when I attempt to open it:
“There are several files with identical file names, which cannot be uniquely assigned.  Affected are:
{{A list of 18 files from “examples"}}

Also, pasting that file outside “examples" gives me this message:
“The selected filename is not importable from the actual project.”

BTW & FYI:  I'm exploring the usage of your software for my fall course that has previously used Logisim in earlier semesters, with Xilinx and Basys3.   With the Covid problems we have in the US, I really now need to make a home-brew lab.  I am exploring something called a “xyLab", a feature made possible from the “Digital Design” xyBook title from xyBooks.com<http://xybooks.com/>.  Actually xyLabs really are intended for their software courses, like “Programming in C" (or C++ Java or Python or Web Programming).  However, this summer the company helped me develop a way to make the zyLab look and grade VHDL code.  The students and I just have to supply the VHDL code, such as generated by your Digital software.

- Dean Johnson
  Department of Electrical and Computer Engineering
  Western Michigan University

hneemann, 2020-07-28:
To make it easier to store a circuit in a VersionControlSystem every circut is stored in a single file. There is no .circ file like in Logisim which stores the whole project with all subcircuits included. 
This means that when you open a file, Digital assumes that all files in the same directory and all subdirectories belong to the current project.
To identify a subcircuit only the file name is used. This makes it easier to reorganize the projects directory structure because you don't need to fix the new path of the subcircuits. This is essentially a copy of the TeX approach.
But this also means that the file names in the project directory and all subdirectories must be unique.

If you store a file outside of the project path, the message "The selected filename is not importable from the actual project." appears, because you can't use the stored circuit in the current project. This is of course perfectly fine if you want to create a new project and do not want to use the saved circuit from the current project.

So the rule of thumb is: If you want to create a new project, first create a new folder and save the necessary files in this folder. If you open a file from that folder, that folder will be the new project folder and everything will work fine. However, this new folder should not be located in a folder used by an older project. Then the older project might not open anymore, because the older project suddenly contains a lot of new files, which might collide with the old existing files.

Hopefully the appearing messages have become a bit clearer now.

closed 2020-07-31 by hneemann