hneemann/Digital #498: Need info on wire colors during simulation

sudhashbahu, 2020-07-29:
[image]

Referring to the attached circuit (a pointless one), I wanted to understand how wire coloring works. When the inputs states are all zeros, the diagram with 1 bit values show a variation of green for values and grey for tri-stated. When input state are all zeros, the diagram with 8 bit values shows blue and grey for tri-stated.

I wanted to understand the logic for the wire colors.

hneemann, 2020-07-29:
This is how the wire color is choosen:

``` Java
public static Style getWireStyle(Value value) {
    if (value == null) return WIRE;
    if (value.isHighZ()) return WIRE_HIGHZ;
    if (value.getBits() > 1) return WIRE;

    if (value.getValue() == 1) return WIRE_HIGH;
    else return WIRE_LOW;
}
```

Where `value==null` means that the simulation is not executed and `value.isHighZ()==true` means that at least one bit of the value is in high-Z state.  In the default color scheme `WIRE` means blue, `WIRE_HIGHZ` gray, and `WIRE_HIGH` and `WIRE_LOW` means bright and dark green.

But now that I think about it, it feels a little inconsistent to me. 
Maybe a bus (`value.getBits() > 1`) should always be displayed in blue, even if it is in high-Z state.

sudhashbahu, 2020-07-29:
Thank you for the clarification. May i suggest  the following:
1. Color is WIRE_HIGHZ when all bit are tri-state
2. Color is WIRE_LOW when all bits are low
3. Color is WIRE_HIGH when all bits are high.
4. Color is WIRE in all other cases.

hneemann, 2020-07-29:
Your suggestion also feels a little weird to me. I have to think about it.

sudhashbahu, 2020-07-29:
The thought behind my suggestion was to be consistent when **all** signals on the bus are of a certain state, irrespective of the "size" of the bus. 

Given that different signals on the bus (greater than 1 bit) can have different states, these scenarios will remain a challenge.

commit 442ae183 (2020-07-30): A data bus is always drawn in the same color, closes #498
closed 2020-07-30 by hneemann