hneemann/Digital #501: Test stats

balbertini, 2020-08-04:
I would like to implement something more verbose than passed/failed to cli testing. It would be useful for me as an automatic assessment to have something like the % of tests that passed. What do you think?

hneemann, 2020-08-05:
In the latest [pre release](https://infdigital.dhbw-mosbach.de/) i have added some more details to the messages.
What do you think about it?

balbertini, 2020-08-05:
I'm using GIT master version and it only shows passed/failed:
```
master$ java -cp target/Digital.jar CLI test -circ ~/Working/sd1/circ_student.dig -tests ~/Working/sd1/circ_golden.dig
Test1: passed
Test2: failed
Tests have failed.
```

The trick that I'm using now for automatic assessment is to split the tests into several small tests and calculating the grade using the output. My proposal is something like (considering only a big test vector):
```
master$ java -cp target/Digital.jar CLI test -circ ~/Working/sd1/circ_student.dig -tests ~/Working/sd1/circ_golden.dig
Test: 90% passed, 10% failed
```
Is there something different on the pre release?

hneemann, 2020-08-05:
> I'm using GIT master version and it only shows passed/failed:

The reason for this is that I made the changes in a local branch. Only the [pre-release build](https://infdigital.dhbw-mosbach.de/) contains them. I don't push all my local branches, which i use to experiment to github.

balbertini, 2020-08-05:
Nice, I downloaded the pre release and its enough for my purposes. Thx

closed 2020-08-05 by balbertini

hneemann, 2020-08-05:
The percentage indicates the percentage of rows in the test case which failed, not the percentage of values which do not match. This seemed most plausible to me.

balbertini, 2020-08-05:
Just for documentation purposes, in case someone is looking for same feature. Using pre release version (see above):
```
prelease$ java -cp target/Digital.jar CLI test -circ ~/Working/sd1/circ_student.dig -tests ~/Working/sd1/circ_golden.dig
Test1: passed
Test2: failed (33%)
Tests have failed.
```