hneemann/Digital #502: I miss click and drag

wmuece, 2020-08-11:
I am enjoying using your digital design software.  I especially find the set of examples to be extremely useful in getting started with any design.  The VHDL export is also very helpful, especially for machine grading of schematic designs.  Therefore, I'm going to use Digital in my freshman digital logic course this semester, abandoning logisim-evolution.

I suspect, however, that the interface will present big problems to students, and I am having some difficulty getting used to it myself. Specifically:
1.  I miss being able to click and drag objects.
2. Also, I am constantly getting "sticky" wires (and objects) dragged around by my pointer by inadvertent clicking.  I wish that new wiring points could be restricted to just clicks on ports and existing wires.

hneemann, 2020-08-11:
I had the same problems using Logisim. That's why I implemented Digital in a way that it was easier to handle for me :-)
But when you have become very familiar with the operation of Logisim, Digital seems unusual. :-)

I like not having to hold down the mouse button while moving a component. Therefore there is no drag when moving components. You just click on them to stick them to the mouse and click again when you have found the new position.
It is the same when drawing wires. You click where you want to attach the wire. If you have reached a pin, or right-clicked, the wire is finished.

wmuece, 2020-08-11:
What about the utility of the click on a point in open space?  That’s the real irritant, because a new “sticky wire” is started. (and it happens constantly) and there really Isn’t a purpose for startIng a wire in open space.

More important question:  I want to make a number of example diagrams showing registers having the traditional long rectangular shape.  These shapes would be used for making processor architecture diagrams, not needed for general design work.  Right now, the “register with enable” is a square shape.  If I want to give it a more rectangular shape, do I need to attach a SVG rectangle to a subcircuited register, or do you have a better suggestion to get my desired shape?  - Dean

Sent from my iPad... Prof. J.

hneemann, 2020-08-11:
I'm not sure I understood exactly what shape you had in mind.
If you insert a register in a subcircuit, you can change the width of this circuit in the circuit specific settings. If you use the layout shape, you can set width and height.

wmuece, 2020-08-11:
Ok, I really just want to change the width of it, that was all.  Thanks for the insight, as I was not aware of this option in circuit specific settings!  - Dean

Sent from my iPad... Prof. J.

hneemann, 2020-08-12:
@wmuece 
> What about the utility of the click on a point in open space?  That’s the real irritant, because a new “sticky wire” is started. (and it happens constantly) and there really Isn’t a purpose for startIng a wire in open space.

Can you remember what your original intention was when you clicked on a point in the open space?

wmuece, 2020-08-12:
Hemut:  I think it was primarily due to my inexperience using the new interface.

I can’t seem to reproduce the cause now.

- Dean Johnson

closed 2020-08-13 by hneemann