hneemann/Digital #506: Documentation on Parameters of a component to use in Generic circuits

sudhashbahu, 2020-08-20:
Section 6 of the documentation (Generic Circuits), contains the text:

_Each parameter has a name and can be modified as an attribute of the field
this. The names of the parameters can be found in the help dialog of the component. If you
want to change the bit width of an adder, the line this.Bits=int(1); can be used._

Contrary to the documentation, the field "Bits" is **not** part of the help dialog. The Help dialog mentions "Data Bits". It is left to the user to figure out that "Data Bits" maps to a filed call "Bits".

hneemann, 2020-08-20:
If the circuit is set to "generic" in the circuit-specific settings, the help dialog of each component lists the keys used to access the attributes.

[image]

sudhashbahu, 2020-08-20:
OK. I see what the problem is:

If you go Help --> Documentation --> 12.1 Adder, you don't see the information

if you go Help --> Components --> Arithmetic --> Adder, you don't see the information

But if adder is in the circuit, then right-click --> Help button ... Does show the proper information.

Looks like it is a consistency issue.

hneemann, 2020-08-20:
Not entirely. I didn't want to add this information everywhere, because 99.9% of the users don't need it.
But I wanted to have it at hand when I edit the generic parameterization. So only the Help button directly below the corresponding dialog field shows this information.