hneemann/Digital #510: Feature Request: Different Color Wires

eric-unc, 2020-09-04:
When a lot of wires in Digital cross each other, it gets pretty confusing. So here's an idea: giving wires a color property. Per default it could be blue, but each wire could be any color you want it to be to make much cleaner circuits.

hneemann, 2020-09-04:
Have you tried the "WIre tool tips" setting?

eric-unc, 2020-09-04:
Oooh interesting, that definitely helps.

closed 2020-09-10 by hneemann

tpitner, 2020-11-03:
Maybe doing something wrongly but I have set the Wire Tool Tips (checked the box) and still cannot see the lines highlighted when mouse hover -- only when selected. The highlighting was not what I meant -- I meant permanent coloring of lines/wires. 

Digital v0.25 on Mac OS Catalina 10.15.7 (19H2), openjdk version "15" 2020-09-15, OpenJDK Runtime Environment (build 15+36-1562), OpenJDK 64-Bit Server VM (build 15+36-1562, mixed mode, sharing)

hneemann, 2020-11-03:
@tpitner 
What about all the other tooltips? Are only the wire tool tips missing, or do components and pins also have none?

tpitner, 2020-11-03:
The hover on components works as expected, ie. display the help when the pointer stops on the component for certain time (maybe 2 sec). On wires, it works under very limited circumstances, i.e. the highlight shows only when the mouse position is exactly on line (not even 1px tolerance) so it is not shown when you move the pointer (even very slowly) across the wire.

hneemann, 2020-11-03:
@tpitner 
Are you using a high resolution retina display?

tpitner, 2020-11-03:
Yes, I use the default resolution which is 2560 x 1600 (MacBook Pro 13") or an external LCD with 2560 x 1440.

hneemann, 2020-11-03:
Could you please zoom in a little and try again? I have found that the catch radius is only half the grid distance. 
If the circuit is zoomed out a lot, that could be too small.

tpitner, 2020-11-03:
It looks you are right -- if I zoomed in & kept the mouse cursor at a wire for a while, I gets highlighted. So it was my fault not keeping the cursor at the wire long enough. Many thanks and sorry for making rumor for no reason :)
Anyway, if the wires can be colored without hovering, it could also be useful.