hneemann/Digital #511: Bug with Bi-directional pins?

mitolegy, 2020-09-11:
there appears to be a bug in the Input component. in previous versions. turning on the settings "is three-state input" and "no zero output" on inputs has allowed for bidirectional data flow through the pin. however, in the latest version. this no longer seems to be true. i kept getting short circuit errors when facing two input components. as if they did not have the two options enabled. 

[image]
basically, turning on these
[image]
used to let me do this.

hneemann, 2020-09-12:
It is more a consitency problem. You also have to set at least one of the inpuits to a default value of Z. Otherwise it could cause a error while starting up the simulation.

But while experimenting to reproduce the bug i found that there is another problem: The default value is sometimes set to "?" which is not allowed. In this case you have to reset the value to "Z".

hneemann, 2020-09-16:
fixed in a526c91bc9a407acd4a764cfccdb94e75d06010f

closed 2020-09-16 by hneemann