hneemann/Digital #512: Finding a Test Signal on a bus

wmuece, 2020-09-14:
Using your 4 bit RC adder example, Digital can find S_i in the test provided, but not when I present S_
i via a bus merger. Any suggestions?  (The test specifically says, "cannot find S_3".)  See attached figures.   We all find this software extremely useful.  Thanks.  - Dean Johnson

[image]
[image]

hneemann, 2020-09-14:
The lower circuit has no output "S_3". If you want to access the first bit of the 4-bit value S, you must access it as in C or Java: `(S>>3)&1`

A complete test of the lower circuit would look like this

```
# complete test of all 512 possible additions
C_in A_3 A_2 A_1 A_0 B_3 B_2 B_1 B_0 C_3 S
loop(A,16)
  loop(B,16)
    let sum = A+B;
0 bits(4,A) bits(4,B) (sum>>4) (sum)
1 bits(4,A) bits(4,B) ((sum+1)>>4) (sum+1)
  end loop
end loop
```

wmuece, 2020-09-14:
H: Thank you.

So this is a signed right shift and I look at the bit on the right end?

What does the &1 say to do?

I noticed that the Merge bus output does not stop the “Export Test Case”, as S3 is known there and Digital enumerates a nice test table anyway.

- Dean

hneemann, 2020-09-15:
The &1 does a AND operation with 1 which isolates the first bit. But this is not necessary if the most significant bit is accessed.

The test cases are generated from a circuit that was previously analyzed. And the analysis can deal with buses.

closed 2020-09-21 by hneemann