hneemann/Digital #514: Underneath component wiring allowed?

wmuece, 2020-09-21:
Not sure if underneath component wiring is allowed, or is this related to my earlier bus output issue?  Some wiring examples follow
[Archive.zip](https://github.com/hneemann/Digital/files/5257028/Archive.zip)

[image]
[image]

hneemann, 2020-09-21:
Underneath component wiring is allowed.
But a pin needs to be placed on a endpoint of a wire. If a pin is placed somewhere at the middle of the wire, the pin is not connected to that wire. If you select and move the two wire endpoints connected to the right subcircuit, you will notice, that they are not connected to the pins of the left subcircuit.

wmuece, 2020-09-21:
Yes, but the issue is how to make a pin.  I couldn’t do it.

Therefore, I made the connections to w and x to each of the MUXes in wired1, which connected fine.  Then I MOVED the connected wires up to make the flat positions under the MUXes to look like Wired2, they suddenly became disconnected.

Not sure how to do this…

-Dean

xotmatrix, 2020-09-22:
See #486

closed 2020-09-28 by hneemann