hneemann/Digital #516: Reset for Register File

ProxyPlayerHD, 2020-09-22:
I recently discovered the beauty of the Register File, which could potentially save me a lot of time constructing a circuit in a RISC CPU.
but it doesn't have a Synchronous Reset to clear all of the Registers at the same time.
the regular Register also doesn't have that but there i can atleast make a Reset with a Multiplexer and an OR Gate.
I cannot do the same for the Register file as i would need to clear each Register separately which would therefore require multiple clock cycles.

so could a Sync. Reset be added to it (and maybe the regular register as well) without breaking synthesizability to Verilog/VHDL?

hneemann, 2020-09-22:
What is the use case when designing a RISC processor? Set all registers to zero in a single cycle? This seems weird to me.

ProxyPlayerHD, 2020-09-22:
basically all CPUs i ever saw had an external Reset pin, so i build a Reset input into every CPU and Circuit i make (if it has internal registers)

so the point would be to implement hardware reset. so that the CPU/Circuit can be set to an intial state at any point by other hardware.

hneemann, 2020-09-22:
But that is not necessary. Just do not rely on the registers being set to zero when programming the boot code. You don't do that with the main memory either. The only things that really need to be reset are the program counter and the IOs.

ProxyPlayerHD, 2020-09-22:
yea that makes sense, i just wanted to be able to reset everything for the sake of completeness.

also while trying to build the circuit with the Register File i ran into some logistic problems and sadly I won't be able to use it as i planned.
It's not related to the Reset thing. It's just that i'm using some of the registers for special purposes that the Register file cannot support and i don't expect it to support because of how specific it is.

so, i'm sorry for wasting both of our time. but i promise i'll find an actual problem to come back with :p
lastly, really liking the dark theme, thanks for that!

closed 2020-09-22 by ProxyPlayerHD

hneemann, 2020-09-22:
> so, i'm sorry for wasting both of our time.

Never mind! :smile: