hneemann/Digital #518: Help to use Digital Simulator

abnerbarros, 2020-09-24:
First of all thank you very much for your excellent work in continuing the Logisim simulator. I am a teacher and have been using Logisim for many years and I missed the features you implemented in your simulator.
I just miss a more detailed tutorial on some features that did not exist in Logisim, such as the use of signal graphics and integration with Verilog icarus.
If you have any kind of description of these features I would very much like to collaborate with the production of the tutorials.
Thank you very much in advance.

abnerbarros, 2020-09-25:
I am encountering a problem when trying to use asynchronous JK FF in a simple counter. Each time I restart the simulation, the FF starts with a different value. This is the first time that I am using this simulator, it probably must be an error on my part, but I didn’t get anything telling me how to ensure that asynchronous FFs are correctly reset at the beginning of the simulation.

hneemann, 2020-09-25:
If you have built your own JK-flopflop, you have to integrate a reset input to bring the circuit into a defined initial state after power-on. A completely symmetrically constructed JK flip-flop does not have a defined initial state.

And even then, a circuit with flipflops that start in a defined way can have an undefined initial state, because edges can appear during the power-on process that change the initial state of a flipflop.

abnerbarros, 2020-09-25:
I realized that the problem only occurs when I invert the clock bit so that it is sensitive to downward transition. When I use the clock in the ascent transition it always works correctly.
In this way all is ok.
[image]
But in this way no.
[image]

abnerbarros, 2020-09-25:
In fact I have now verified that the problem occurs in both cases, only that it is more frequent when the clock is inverted. The circuit really does not seem to respond correctly reset proccess, each time I restart it is in a different state.

[image]
[image]
All FF is configured to star with 0.

abnerbarros, 2020-09-25:
I identified the problem and found a palliative solution.
The problem occurs when starting the clock component in manual mode. Apparently it is generating glitches, which causes FF to change states.
The solution was to ensure that these pulses were not propagated. However, for me the best way would be to ensure that the FFs remain in the reset state waiting until the simulation is fully started

[image]

hneemann, 2020-09-25:
Your observations are correct, but it is not a bug. It behaves exactly as it should.
The JK flip-flops are completely independent of each other. There is no hidden logic that initializes the jk-flipflops with a synchronous reset pulse. To provide such a circuit with a stable initial state, a reset circuitry must be added.
Or you build a synchronous counter. That would certainly be the more elegant way.

abnerbarros, 2020-09-25:
I work in my discipline with synchronous and asynchronous accountants, so I need to present both models to my students.
With regard to FF, it is expected that those that do not have asynchronous inputs will have some type of initialization logic, also because the component itself has the option of indicating an initial state. Anyway it seems to me that my problem was not due to the FFs, but to the clock component in manual mode.

hneemann, 2020-09-26:
I would think that the clock component has nothing to do with it. 
If J and K are held at zero during startup, a glitch at the C input has no effect.

closed 2020-09-29 by hneemann