hneemann/Digital #520: Help tu use pause component

abnerbarros, 2020-09-25:
How to use the pause component? I inserted the component in my project but the Run to Break button is not being activated.

[image]

hneemann, 2020-09-26:
The break component is used to clock the circuit forward to a certain condition. It is not intended to be used if the real time clock is activated. The intended usage is to start the circuit in "non real-time mode" and then instead of clocking the circuit forward click by click to the condition you want to investigate, you are able to clock the circuit forward to this condition automatically.

The stop component simply stops the simulation like the stop button does.

abnerbarros, 2020-09-27:
It stay confuse for me yet, may you send me one example?

hneemann, 2020-09-27:
Open the attached circuit:

At first:
1. Start the simulation, the "Run to Break" button is enabled. (Directly to the right of the Start button.)
2. Set en to high.
3. Clock the circuit to a rising edge of Q3 by clicking on the clock component multiple times.

Now:
1. Start the simulation again.
2. Set en to high.
3. Click on the "Run To Break" button.

[cntBrk.dig.zip](https://github.com/hneemann/Digital/files/5287877/cntBrk.dig.zip)

abnerbarros, 2020-09-27:
This is the problem, the "Run to Break" button   and F7 command always stay disabled for me. 
[image]
I just open your file and make this printscreen.

abnerbarros, 2020-09-27:
Great, now it wok fine. Thanks so much.

closed 2020-09-28 by hneemann